osrf/gazebo_tutorials #108: When I use set_velocity plugin make error

richardsanqi, 2020-06-02:
In file included from /home/lipi/Documents/set_vel_plugin/src/SetLinkVelocityPlugin.cpp:21:0:
/home/lipi/Documents/set_vel_plugin/include/ode_perfect_velocity.hh: In member function ‘void gazebo::OdePerfectVelocityController::Start(gazebo::physics::LinkPtr, gazebo::math::Vector3, gazebo::math::Vector3, double, double)’:
/home/lipi/Documents/set_vel_plugin/include/ode_perfect_velocity.hh:36:36: error: ‘class gazebo::physics::Model’ has no member named ‘CreateLink’
         this->phantomLink = model->CreateLink("__perfect_phantom_link__");
                                    ^
CMakeFiles/SetLinkVelocityPlugin.dir/build.make:62: recipe for target 'CMakeFiles/SetLinkVelocityPlugin.dir/src/SetLinkVelocityPlugin.cpp.o' failed
make[2]: *** [CMakeFiles/SetLinkVelocityPlugin.dir/src/SetLinkVelocityPlugin.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/SetLinkVelocityPlugin.dir/all' failed
make[1]: *** [CMakeFiles/SetLinkVelocityPlugin.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I follow the gazebo tutorials,and use gazebo7.It seems cmake works without error,but when i use make ,the terminal get this error,ubuntu 16.06+ros kinetic

chapulina, 2020-06-02:
I believe you're talking about this tutorial:

http://gazebosim.org/tutorials?tut=set_velocity

You may be using an old Gazebo 7 version. That function [was added on 7.1.0](https://github.com/osrf/gazebo/blob/gazebo11/Changelog.md#gazebo-710-2016-04-07). Update your version as explained [on this tutorial](http://gazebosim.org/tutorials?cat=install&tut=install_ubuntu&ver=7.0).

richardsanqi, 2020-06-02:
> I believe you're talking about this tutorial:
>
> http://gazebosim.org/tutorials?tut=set_velocity
>
> You may be using an old Gazebo 7 version. That function [was added on [...]

Thanks a lot.I just reinstall the GAZEBO,and this problem is sloved.But I use this command GAZEBO_PLUGIN_PATH=. gazebo --pause --verbose ../set_velocity.world as tutorials,the model is loaded,but they don not move.In fact，I have no idea how to use this plugin.Although the tutorials give some code.I want to set my robot jointvelocity real-time by using this plugin.Do you have any program or example to set robot jointvelocity.Thanks！

bai-ivan, 2023-03-11:
I have a bit of hacky way of fixing this. I know it's old issue, but here it is for everyone else that might be googling.
DISCLAIMER: The hacky part modifies configurations on your system, modify it only if you understand what can happen later.

Code in this repo is probably a bit old, and there's stuff in gazebo dependencies that changed. Since i need only joint velocity example, you can completely ignore all the errors from the link one. In case you need the link example - code is a bit too broken for me to fix it at the moment, so you're on your own.

To get the example to build, edit your gazebo's cmake config.
```bash
sudo nano /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake
```

find the line (mine was 62, but i believe it may vary) where it says ```set (GAZEBO_HAS_SIMBODY TRUE)``` and change it to ```set (GAZEBO_HAS_SIMBODY FALSE)```

then open the cmake-lists of the example (this repo)
```bash
nano /path/to/gazebo_tutorials/set_velocity/examples/set_vel_plugin/CMakeLists.txt
``` 
and remove the lines 
```bash
add_library(SetLinkVelocityPlugin SHARED src/SetLinkVelocityPlugin.cpp)
target_link_libraries(SetLinkVelocityPlugin ${GAZEBO_LIBRARIES})
```
Save the CMakeLists.txt, clear the build cache from the build folder, and start the cmake command again. Voila.