osrf/gazebo_tutorials #131: Target "ignition-common1::ignition-common1" not found for gazebo-9 on windows10

ewillsbae, 2020-12-04:
I'm trying to build gazebo9 on Windows10, using VS2019. I'm following the usual tutorial (http://gazebosim.org/tutorials?tut=in...). I've made it through every part so far with no issues, and now I'm trying to build Gazebo itself.

After running ..configure.bat, I get the following errors:

-- Found Windows SDK Universal CRT include directory: C:/Program Files (x86)/Windows Kits/10/Include -- Configuration successful. Type make to compile gazebo -- Configuring done CMake Error at gazebo/gui/CMakeLists.txt:212 (target_link_libraries): Error evaluating generator expression: $<TARGET_PROPERTY:ignition-common1::ignition-common1,INTERFACE_INCLUDE_DIRECTORIES> Target "ignition-common1::ignition-common1" not found. CMake Error: Error evaluating generator expression: $<TARGET_PROPERTY:ignition-common1::ignition-common1,INTERFACE_INCLUDE_DIRECTORIES> Target "ignition-common1::ignition-common1" not found.

This repeats for just about any gazebo/____/CMakeLists.txt.

Ignition common was built successfully and found by other ign-____ packages that needed it when compiling. Can anyone suggest where to fix this? I've tried linking and including the path to it in the various configure.bat file with no luck either.

j-rivero, 2020-12-09:
The documentation and artifacts to compile on Windows are outdated. Instead of fixing the current mess, let's try to find a better approach, we'll appreciate your experience and opinions on the alternatives proposed and linked in https://github.com/osrf/gazebo/issues/2901. Thanks.

closed 2020-12-09 by j-rivero

ewillsbae, 2020-12-09:
This is fixed by using vcpkg to install the ignition packages: https://ms-iot.github.io/ROSOnWindows/Build/rosdeps_vcpkg.html#references

j-rivero, 2020-12-09:
> This is fixed by using vcpkg to install the ignition packages: https://ms-iot.github.io/ROSOnWindows/Build/rosdeps_vcpkg.html#references

Thanks for the pointer. All add it to the list of alternatives.