osrf/gazebo_tutorials #139: "No TF data" in rviz when running ros_depth_camera tutorial

FSund, 2021-02-25:
I'm trying to follow the [ros_depth_camera](https://github.com/osrf/gazebo_tutorials/blob/master/ros_depth_camera/tutorial.md) tutorial, but having some issues when trying to visualize in RViz.

I launch gazebo via `roslaunch gazebo_ros empty_world.launch`, add my model and run `rosrun rviz rviz`, but I get "No TF data" under Global Status/Fixed Frame, so I'm unable to visualize the PointCloud2.

Environment:
- Ubuntu 20.04
- ROS Noetic

`model.sdf`

```
<?xml version="1.0" ?>
<sdf version="1.5">
  <model name="kinect_depth">
    <static>true</static>
    <pose>0 0 0.036 0 0 0</pose>
    <link name="link">
      <inertial>
        <mass>0.1</mass>
      </inertial>
      <collision name="collision">
        <geometry>
          <box>
            <size>0.073000 0.276000 0.072000</size>
          </box>
        </geometry>
      </collision>
      <visual name="visual">
        <geometry>
          <mesh>
            <uri>model://kinect_depth/meshes/kinect.dae</uri>
          </mesh>
        </geometry>
      </visual>
      <sensor name="camera" type="depth">
        <update_rate>20</update_rate>
        <camera>
          <horizontal_fov>1.047198</horizontal_fov>
          <image>
            <width>640</width>
            <height>480</height>
            <format>R8G8B8</format>
          </image>
          <clip>
            <near>0.05</near>
            <far>3</far>
          </clip>
        </camera>
        <plugin name="camera_plugin" filename="libgazebo_ros_openni_kinect.so">
          <baseline>0.2</baseline>
          <alwaysOn>true</alwaysOn>
          <!-- Keep this zero, update_rate in the parent <sensor> tag
            will control the frame rate. -->
          <updateRate>0.0</updateRate>
          <cameraName>camera_ir</cameraName>
          <imageTopicName>/camera/color/image_raw</imageTopicName>
          <cameraInfoTopicName>/camera/color/camera_info</cameraInfoTopicName>
          <depthImageTopicName>/camera/depth/image_raw</depthImageTopicName>
          <depthImageCameraInfoTopicName>/camera/depth/camera_info</depthImageCameraInfoTopicName>
          <pointCloudTopicName>/camera/depth/points</pointCloudTopicName>
          <frameName>camera_link</frameName>
          <pointCloudCutoff>0.5</pointCloudCutoff>
          <pointCloudCutoffMax>3.0</pointCloudCutoffMax>
          <distortionK1>0</distortionK1>
          <distortionK2>0</distortionK2>
          <distortionK3>0</distortionK3>
          <distortionT1>0</distortionT1>
          <distortionT2>0</distortionT2>
          <CxPrime>0</CxPrime>
          <Cx>0</Cx>
          <Cy>0</Cy>
          <focalLength>0</focalLength>
          <hackBaseline>0</hackBaseline>
        </plugin>
      </sensor>
    </link>
  </model>
</sdf>
```

martinerk0, 2021-04-14:
You need to type **camera_link** in Fixed Frame in Global options.

FSund, 2021-04-14:
That is obvious to me now that I have some more experience with ROS, but it was not clear from the tutorial at the time. Perhaps a notice about this should be added to the tutorial?

martinerk0, 2021-04-14:
I agree. Also why is camera in rviz pointing upwards? There are 5 Q/A hacks doing multiple links / using static transform but people doing tutorials should not google this for hours.