osrf/gazebo_tutorials #151: CMake Error when compiling set_vel_plugin

muronglindong, 2021-07-26:
Hello, I'm learning Setting Velocity on Links And Joints, but when I downloaded all the codes and compiled the set_vel_plugin as the tutorials guiding, there exists a cmake error:
`mu@mu-TM1703:~/gazebo_tutorials-master/set_velocity/examples/set_vel_plugin/build$ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/simbody/SimbodyConfig.cmake:131 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:160 (find_package)
  CMakeLists.txt:4 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found ccd: /usr/include (found suitable version "2.0", minimum required is "2.0") 
-- Found fcl: /usr/include (found suitable version "0.5.0", minimum required is "0.3.2") 
-- Found assimp: /usr/include (found version "5.0.0") 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Looking for ignition-math6 -- found version 6.8.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.8.0
-- Looking for ignition-transport8 -- found version 8.2.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.8.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Looking for ignition-common3 -- found version 3.13.1
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE  
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.13.1
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.8.0
-- Looking for ignition-fuel_tools4 -- found version 4.3.0
-- Searching for dependencies of ignition-fuel_tools4
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.68.0")  
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Found JSONCPP: TRUE  
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Found YAML: TRUE  
-- Checking for module 'libzip'
--   Found libzip, version 1.5.1
-- Found ZIP: TRUE  
-- Looking for ignition-common3 -- found version 3.13.1
-- Looking for ignition-msgs5 -- found version 5.7.0
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Simbody_LIBRARIES (ADVANCED)
    linked by target "SetJointVelocityPlugin" in directory /home/mu/gazebo_tutorials-master/set_velocity/examples/set_vel_plugin
    linked by target "SetLinkVelocityPlugin" in directory /home/mu/gazebo_tutorials-master/set_velocity/examples/set_vel_plugin

-- Configuring incomplete, errors occurred!
See also "/home/mu/gazebo_tutorials-master/set_velocity/examples/set_vel_plugin/build/CMakeFiles/CMakeOutput.log".
See also "/home/mu/gazebo_tutorials-master/set_velocity/examples/set_vel_plugin/build/CMakeFiles/CMakeError.log".
`

How to solve this problem?
 I use ubuntu 20.04, gazebo 11, thank you!

j-rivero, 2021-07-27:
Seems to me a problem related to the installation of Simbody libraries. Could you please remove them (`apt-get remove .*simbody.*`, install them again (`apt-get install libsimbody-dev`), remove your CMakeCache.txt file and try to  run cmake again to check if it is a problem on your local system?

MyDearestJingo, 2022-08-02:
> Seems to me a problem related to the installation of Simbody libraries. Could you please remove them (`apt-get remove .*simbody.*`, install them again (`apt-get install libsimbody-dev`), remove your CMakeCache.txt file and try to run cmake again to check if it is a problem on your local system?

I got the same error, but nothing changed after reinstalling libsimbody-dev (`libsimbody-dev is already the newest version (3.6.1+dfsg-7build1)`)